mmurdoch/arduinounit #44: Add tolerance for double comparison

DanNixon, 2016-01-17:
Doubles should really be compared using a tolerance, this is my current work around:

```cpp
void assertClose(double a, double b, double tol = 0.01)
{
  assertTrue(abs(a - b) < tol);
}
```

wmacevoy, 2016-04-06:
Try branch iss44 and see if you like the version.  I created an assertClose which allows for relative and absolute errors, and prints the arguments if the assertion fails.  Note assertEqual is not changed.

wmacevoy, 2016-04-06:
I think your macro would be better as

assertLess(abs(b-a),tol)

because it would report the error amount if it failed.

wmacevoy, 2016-08-11:
I'm closing this issue, the assertLess above works well and avoids assumptions about tolerance.

closed 2016-08-11 by wmacevoy