mmurdoch/arduinounit #45: Fail to compile RFDuino

pushtheworldllc, 2016-03-31:
Failling to compile on RFDuino with error:

```
/Applications/Arduino_1.5.8.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: address 0x20004374 of /var/folders/_w/wvsmn93n26q5v65gyt01x3p40000gn/T/build5167854416226854633.tmp/OpenBCI_Radio_Test.cpp.elf section `.bss' is not within region `ram'
/Applications/Arduino_1.5.8.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: address 0x20004374 of /var/folders/_w/wvsmn93n26q5v65gyt01x3p40000gn/T/build5167854416226854633.tmp/OpenBCI_Radio_Test.cpp.elf section `.bss' is not within region `ram'
/Applications/Arduino_1.5.8.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: address 0x20004374 of /var/folders/_w/wvsmn93n26q5v65gyt01x3p40000gn/T/build5167854416226854633.tmp/OpenBCI_Radio_Test.cpp.elf section `.bss' is not within region `ram'
collect2: error: ld returned 1 exit status
Error compiling.
```

wmacevoy, 2016-04-01:
Sorry we do not test on this platform; it seems there is a configuration problem with the compiler or you are out of memory on the target platform.

Warren MacEvoy
wmacevoy@coloradomesa.edu
http://www.coloradomesa.edu/~wmacevoy
970-248-1070

pushtheworldllc, 2016-04-01:
can we add it?

wmacevoy, 2016-04-01:
A little searching strongly suggests this is the error you get (section `.bss' is not within region `ram') when you have over-allocated ram.  If your test suite is kind of big; try commenting out the test and noting how much space is left without testing.  I think you are near your system limits.  Adding the platform to our tests wont change this.  Take a look at https://github.com/mmurdoch/arduinounit/issues/30 for how to block group your tests (I'd make sure the sketch works without tests and how much room is left before trying to fit them in; you may have to test components in parts and leave the tests out of the final firmware)

wmacevoy, 2016-08-11:
I'm closing this issue because it is not a principle arduino platform.

closed 2016-08-11 by wmacevoy