mmurdoch/arduinounit #46: Order of tests

sidey79, 2016-04-06:
Hi,

more a question than a request:
Is there a way to specify the order of the Tests?

I have some tests, which needs to be run after another test.
Can I control the order in a way other than assigning numbers to the test names ?

wmacevoy, 2016-04-06:
The tests (that are not skipped) are run in alphabetical order.  (so strcmp(before,after) < 0).  For continuous tests (testing(xxx) not test(xxx)) you can check if another test is completed before starting yours.  These are the checkTestDone(name) macros.

sidey79, 2016-04-06:
sorry, it is not clear to me, how to to it.

Assume i have three tests (first, second, third) and on testing

testing(mastertest)
{
	if (checkTestDone(first))
	{
		if (checkTestDone(second))
		{
			checkTestDone(third);
			pass();
		}
	}
}

The tests are run in the alphabetical order. No idea, how to change that with the testing.

wmacevoy, 2016-04-07:
https://gist.github.com/wmacevoy/f2cbd5d5761dd7eecb66b5de8dc9e20a

or

https://gist.github.com/wmacevoy/36bd8fa7660a20e1adc225c0e8bf12e9

closed 2016-06-26 by sidey79