mmurdoch/arduinounit #47: Bug while comparing character arrays

nehchal, 2016-04-25:
The assertEqual() doesn't seem to work with character arrays. Following code fails the test:
```
test(string)
{
  char str1[32];
  char *str2="test1";
  strcpy(str1,str2);
  assertEqual((char*)str1,str2); // ok --  
} 
```
Error message is as follows:
```
Assertion failed: ((char*)str1=test1) == (str2=test1), file /tmp/build91e6c79303c447d1075081849910176c.tmp/sketch/tests/Utils_test.cpp, line 31.
Test string failed.
Test summary: 0 passed, 1 failed, and 0 skipped, out of 1 test(s).
```
I think, assertEqual is comparing pointer values (the addresses) instead of parsing the whole string and comparing character by character (which something like strcmp() would do). I have worked around it by using: assertEqual( strcmp(str1, str2), 0);

wmacevoy, 2016-05-23:
Const char * works (which is what I tested but did not write in the readme).  That AU works badly for char*, char[], const char * has been a nagging bug for me.  [ it DOES work if you cast to const char *,
but beginners should not have to care).  Please try branch iss47 which should work with no casting.

I will update the readme for the master branch to reflect that the current version requires the const char * conversion.

wmacevoy, 2016-08-11:
The char* vs const char * is reserved as a future enhancement; the readme now reflects proper use of the comparison.

closed 2016-08-11 by wmacevoy