mmurdoch/arduinounit #48: WARNING: Category '' in library ArduinoUnit is not valid. Setting to 'Uncategorized'

gokhandilek, 2016-05-16:
Getting this error in my arduino IDE 1.6+.
Anyone knows what it might be?

wmacevoy, 2016-05-23:
I think the library.properties file was in the wrong folder (it should have been in the src folder).  Copy the src/library.properties file to your library/ArduinoUnit folder and it should go away.

closed 2016-08-13 by gokhandilek

kostrunx, 2018-06-14:
There is no line "category=???" in your's "library.properties" file, that is in every subfolder of the "libraries" folder. Just find the appropriate subfolder ("ArduinoUnit" in your case), and add the missing line. Be sure to replace the ??? with the corect category. For more info take a look at:
[https://github.com/arduino/arduino/wiki/arduino-ide-1.5:-library-specification](url)