mmurdoch/arduinounit #51: Run all tests from setup

tgm-git, 2016-11-21:
I have attempted to use continuous integration during this project and therefore I need to run a series of unit tests as often as possible. ArduinoUnit has been a good framework for creating and running unittests, but I would like to be able to call them once from the end of setup.

The idea being that any time I run my code, I run the tests too. If they are run from loop they will remain in RAM and they are continuously checked if they have already run. This is not ideal, I would prefer that they are run in setup and then removed from RAM leaving more space for the project code to run.

The idea is to automate the process. At the moment I can either run the tests or my project code, but not both due to the heavy RAM usage of the project code.

wmacevoy, 2016-11-23:
This is a very complex request; the normal deployment of an arduino app is
a fixed firmware uploaded once.  You are asking for an overlay/bootloader
that has phases and an alternative storage for additional code.  This just
falls outside the scope of arduinounit.  I agree space for testing is an
issue in an embedded design; see
https://github.com/mmurdoch/arduinounit/issues/30 for a discussion on
limiting image size using conditional compilation.  I don't know of a
project that helps manage firmware overlays; I suggest using a larger
system for testing that can accommodate the tests and application together
as one image.

soundanalogous, 2016-11-23:
One other technique that we're working on with the Firmata library that may be a useful example is to extract as much business logic as possible out of any code that needs to run on the board into classes that can run in any c++ environment. See Firmata.cpp, FirmataParser.cpp and FirmataMarshaller.cpp [here](https://github.com/firmata/arduino). All of this code used to be in Firmata.cpp, but by extracting the logic into c++ classes without any Arduino dependencies those classes can now be tested in a desktop (or even server) environment. In our case the tests for the Parser and Marshaller classes have not been written yet as it was more of a realization after the refactoring was done. But that is the plan going forward.

wmacevoy, 2016-12-13:
Agree that moving unit tests (when appropriate) off the embedded hardware is better; but can be hard for subtle and not-so-subtle reasons.  The sizes of ints/doubles, support of dynamic structures, and memory size and layout (ie flash and EEPROM) are all not so easy to write portable tests around.  Getting a mock arduinounit framework so tests can be run away from the embedded hardware would be an interesting goal (that nobody has asked for yet).

By the way, testing(feature) tested on each loop until resolved (pass(), fail(), skip()), and test(feature) tests are tested once.  Once tests are resolved, they are removed from the list of active tests (which eventually becomes empty once all tests are resolved).  So the Test::run() in loop() does not waste cycles on resolved tests.

closed 2016-12-13 by wmacevoy