mmurdoch/arduinounit #53: assertEqual rounding problems - caused by Arduino?

mathieub93, 2017-01-10:
_working with gyroscope and accelerometer trying to triangulate actual position._

So I was writing the following test in Arduino and I got to an issue.

```
double degreeToRad = PI / 180;

// A = cos θ * H
test(mf12t1) {
  double in1 = 60;
  double in2 = 2;
  double out = 1;
  double res = cos(in1 * degreeToRad) * in2;
  assertEqual(out, res);
}
```
The issue I came across was: The output I should have and the result from the formula were both 1.00, but the test failed anyway.
I started inspecting the error and figured out that res is actually something like `0.9999999046` due to rounding errors.

Because of this error I wrote a little piece of code to fix it for myself.

```
void assertAlmostEqual(double out, double res)
{
  double m = 1000.0;
  assertEqual(round(out * m) / m, round(res * m) / m);
}
```

I don't know if this is something; Arduino should fix, could be fixed in this library, or just has to be worked around.
But I thought, since I found the 'error' I could atleast mention it.

mmurdoch, 2017-01-10:
Rounding problems like these are due to the underlying way floating point arithmetic is handled. Other unit testing frameworks work around these kinds of rounding problems by having a dedicated `assertEquals()` function which takes an additional `delta` parameter to describe the maximum amount of rounding error which is acceptable.

For example, [from jUnit](http://junit.org/junit4/javadoc/latest/org/junit/Assert.html):

> `public static void assertEquals(float expected, float actual, float delta)`
>
> Asserts that two floats are equal to within a positive delta. If they are not, an AssertionError is
> thrown. If the expected value is infinity then the delta value is ignored. NaNs are considered equal: `assertEquals(Float.NaN, Float.NaN, *)` passes.
> Parameters:
> `expected` - expected value
> `actual` - the value to check against expected
> `delta` - the maximum delta between expected and actual for which both numbers are still considered equal.

Perhaps we should probably add a similar function to handle this?

wmacevoy, 2017-01-26:
This came up before as issue44, and there is a branch iss44 which incorporated a version of this.  assertCllose().  In the end I didn't get any feedback on it and there is always the complication of what you mean by "close" (sometimes relative, sometimes absolute error).  Without these, I'm not sure how this is better than assertLess(abs(x-y),eps), which is why I closed that issue without incorporating it into the master branch.

closed 2017-07-29 by wmacevoy