mmurdoch/arduinounit #54: Missing pgmspace.h

BlackEdder, 2017-01-30:
I am trying to use this library through platformio on my nodemcuv2. When I try to compile it I got the following error:
```
Compiling .pioenvs/nodemcuv2/lib/ArduinoUnit_ID946/ArduinoUnitUtility/FakeStreamBuffer.o
In file included from .piolibdeps/ArduinoUnit_ID946/src/ArduinoUnit.h:38:0,
from .piolibdeps/ArduinoUnit_ID946/src/ArduinoUnitUtility/ArduinoUnit.cpp:2:
Compiling .pioenvs/nodemcuv2/lib/ArduinoUnit_ID946/ArduinoUnitUtility/FreeMemory.o
.piolibdeps/ArduinoUnit_ID946/src/ArduinoUnitUtility/Compare.h:3:26: fatal error: avr/pgmspace.h: No such file or directory
#include <avr/pgmspace.h>
^
compilation terminated.
*** [.pioenvs/nodemcuv2/lib/ArduinoUnit_ID946/ArduinoUnitUtility/ArduinoUnit.o] Error 1
```

pacm93, 2017-02-20:
Same here

`Arduino: 1.8.1 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/project/runner2/runner2.ino:7:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

wmacevoy, 2017-02-20:
I do not have this board to see what is wrong.  If you look at the Compare.h header file, it only includes avr/pgmspace.h if the F macro is defined, which is designed to move literal strings into flash instead of ram.  This has been true only for AVR chips which support flash and therefore have the avr/pgmspace headerfile (needed to implement movement of data between flash and ram). 

It seems like, on your platform, the F macro is defined but the avr/pgmspace is not.  Does it support flash literals with the F macro; and how do you get the comparison equivalents strcmp_P, memcpy_P?  If you can determine the architecture with a proper if defined(...) at the top of Compare.h and include the corresponding header (or define an equivalent) then this can be built.

Without the board or dev tools, this is all the help I can give....

pacm93, 2017-02-22:
Hi, thanks for the feedback. The F macro is defined for my board. I tried to include the pgmspace.h from the ESP8266 library into compare.h like this

```
#if defined(F)
//#include <avr/pgmspace.h>
#include "/home/pacm/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/pgmspace.h"
#endif
```

Even so, I am getting this error:

```
Arduino: 1.8.1 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from /home/pacm/arduino-jenespi/runner2/runner2.ino:7:0:
/home/pacm/arduino-jenespi/runner2/runner2.ino: In member function 'virtual void test_ok::once()':
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:575:77: error: 'typeof' was not declared in this scope
 #define assertOp(arg1,op,op_name,arg2) do { if (!Test::assertion<typeof(arg1),typeof(arg2)>(F(__FILE__),__LINE__,F(#arg1),(arg1),F(op_name),op,F(#arg2),(arg2))) return; } while (0)
                                                                             ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:578:38: note: in expansion of macro 'assertOp'
 #define assertEqual(arg1,arg2)       assertOp(arg1,compareEqual,"==",arg2)
                                      ^
/home/pacm/arduino-jenespi/runner2/runner2.ino:18:3: note: in expansion of macro 'assertEqual'
   assertEqual(x,y);
   ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:575:159: error: no matching function for call to 'test_ok::assertion(const __FlashStringHelper*, int, const __FlashStringHelper*, int&, const __FlashStringHelper*, <unresolved overloaded function type>, const __FlashStringHelper*, int&)'
 #define assertOp(arg1,op,op_name,arg2) do { if (!Test::assertion<typeof(arg1),typeof(arg2)>(F(__FILE__),__LINE__,F(#arg1),(arg1),F(op_name),op,F(#arg2),(arg2))) return; } while (0)
                                                                                                                                                               ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:578:38: note: in expansion of macro 'assertOp'
 #define assertEqual(arg1,arg2)       assertOp(arg1,compareEqual,"==",arg2)
                                      ^
/home/pacm/arduino-jenespi/runner2/runner2.ino:18:3: note: in expansion of macro 'assertEqual'
   assertEqual(x,y);
   ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:575:159: note: candidate is:
 #define assertOp(arg1,op,op_name,arg2) do { if (!Test::assertion<typeof(arg1),typeof(arg2)>(F(__FILE__),__LINE__,F(#arg1),(arg1),F(op_name),op,F(#arg2),(arg2))) return; } while (0)
                                                                                                                                                               ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:578:38: note: in expansion of macro 'assertOp'
 #define assertEqual(arg1,arg2)       assertOp(arg1,compareEqual,"==",arg2)
                                      ^
/home/pacm/arduino-jenespi/runner2/runner2.ino:18:3: note: in expansion of macro 'assertEqual'
   assertEqual(x,y);
   ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:493:17: note: template<class A, class B> static bool Test::assertion(const __FlashStringHelper*, uint16_t, const __FlashStringHelper*, const A&, const __FlashStringHelper*, bool (*)(const A&, const B&), const __FlashStringHelper*, const B&)
     static bool assertion(const __FlashStringHelper *file, uint16_t line, const __FlashStringHelper *lhss, const A& lhs, const __FlashStringHelper *ops, bool (*op)(const A& lhs, const B& rhs), const __FlashStringHelper *rhss, const B& rhs) {
                 ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:493:17: note:   template argument deduction/substitution failed:
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:575:159: error: template argument 1 is invalid
 #define assertOp(arg1,op,op_name,arg2) do { if (!Test::assertion<typeof(arg1),typeof(arg2)>(F(__FILE__),__LINE__,F(#arg1),(arg1),F(op_name),op,F(#arg2),(arg2))) return; } while (0)
                                                                                                                                                               ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:578:38: note: in expansion of macro 'assertOp'
 #define assertEqual(arg1,arg2)       assertOp(arg1,compareEqual,"==",arg2)
                                      ^
/home/pacm/arduino-jenespi/runner2/runner2.ino:18:3: note: in expansion of macro 'assertEqual'
   assertEqual(x,y);
   ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:575:159: error: template argument 2 is invalid
 #define assertOp(arg1,op,op_name,arg2) do { if (!Test::assertion<typeof(arg1),typeof(arg2)>(F(__FILE__),__LINE__,F(#arg1),(arg1),F(op_name),op,F(#arg2),(arg2))) return; } while (0)
                                                                                                                                                               ^
/home/pacm/arduino-jenespi/libraries/arduinounit-2.2.0/src/ArduinoUnit.h:578:38: note: in expansion of macro 'assertOp'
 #define assertEqual(arg1,arg2)       assertOp(arg1,compareEqual,"==",arg2)
                                      ^
/home/pacm/arduino-jenespi/runner2/runner2.ino:18:3: note: in expansion of macro 'assertEqual'
   assertEqual(x,y);
   ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

wmacevoy, 2017-02-22:
Looking at https://github.com/neu-rah/ArduinoMenu/issues/68, I think you want something like
```c
   #ifdef ESP8266
   #include <esp8266/pgmspace.h>
   #define typeof(x) __typeof__(x)
   #endif
```

crankyoldgit, 2017-05-08:
FYI, I think I have a solution to the ESP8266 issue in #57

mahyarPourjabbar, 2017-12-09:
updating the Arduino AVR Boards from Arduino software will entirely solve the problem!
[image]

crankyoldgit, 2017-12-09:
@mahyarPourjabbar I don't think it will, as the ESP8266 is not an "Arduino AVR Board". But hey, if they do support it now, awesome.

wmacevoy, 2018-04-04:
[thread summary] ArduinoUnit 2.3.5-alpha now supports flash string de-duplication on AVR and uses RAM on other platforms, along with a number of memory-saving optimizations that do not change the outcome of tests. It also supports ESP 8266 and ESP 32 platforms (along with development platform "en vitro" tests). Please try the latest pre-release and provide feedback on #73 if you are interested in these features.

closed 2018-04-04 by wmacevoy