mmurdoch/arduinounit #61: Running tests multiple times

matthijskooijman, 2017-07-26:
I just got started with ArduinoUnit and it has been serving me pretty well so far. Well-thought out API and implementation, kudos!

However, after adding a few tests I'm running into a limitation: I want to run some or all tests multiple times, but with different options. It seems that ArduinoUnit pretty much makes this impossible without reimplementing some parts of the framework (e.g. parts of `Test::run()`, `Test:resolve()`, etc.).

I'm not really sure what the best way would be to support this, but it seemed good to discuss things here first. Usecases I'm thinking about:
 - Simply re-running all tests, but with different settings (e.g. set through global variables or members on the tests themselves). This is prevented because `Test::run()` removes them from the list, so afterwards there is no way to access the full list of tests (needing to manually iterate them) and there is no way to re-insert the tests into the list (`root` and `insert()` are private). Such a re-run should probably show another summary.
 - Running a selection of tests, possibly in a different order. This would require explicitly listing the tests. Currently you just call `loop()` and `resolve()` on them manually, but that does not show a nice summary (or shows it every time if the list is already empty).
 - Combining the previous two: Run specific tests first, then run all others.

Writing this, I think that some changes that would be needed would be to:
 - Add an option to keep tests in the list after they complete.
 - Add a `resetAll()` method or similar, that resets all tests (that are still in the list) to `UNSETUP` (a test that only wants to run setup once and not re-run it in reset can keep its own flag?). This method could also call a virtual `reset()` on each test for more control. The `resetAll()` method should also reset the counters.
 - Add a `run(Test*)` method that runs the loop for just that test (e.g. the core of the current `Test::run()` without the loop.
 - Add a `showSummary()` option to force the summary to be shown.
 - Add an option to prevent the summary from being run when all tests have run (so you can re-run some tests)

The last points wrt the summary are a bit more tricky - How does a test count when it is run twice, or not at all? What about tests there were started but not stopped?

How does this sound?

wmacevoy, 2017-07-27:
A test(xxx) test is run once.  A testing(xxx) test is run until pass, fail or skip (once per Test::run()).  If you have multiple options, you might write it like

```
void myTest(int option) {
 // since this is a regular function, you have to use 
 //  Test::pass(), Test::fail(), etc., since the compiler
 //  does not know this is part of a test.
}

testing(op1) { myTest(1); }
testing(op2) { myTest(2); }

```

You can also extend the Test class if you want a more structured test (extend TestOnce if you only want the test to run once).  That might look like:

```
class MyTest : public Test {
public:
  int option;
  MyTest(const __FlashStringHelper *name, int _option) : Test(name), option(_option) {
    // TODO: construct named test.
    // This should be lightweight - it may be excluded
    //
    // You can set verbosity.
  }
  void setup() {
    // TODO: setup test
    // You can call pass(), fail(), or skip() to immediately resolve test
    // You can make assertions.
    // You can set verbosity.
  }
  void loop() {
    // TODO: run test on each loop
    // You can call pass(), fail(), or skip() to resolve test
    // You can make assertions.
    // You can set verbosity.
  }
};

MyTest myTest1(F("myTest1"),1);
MyTest myTest2(F("myTest2"),2);
```
If you want the test to only run once, then extend TestOnce instead.  This is more typing but very flexible, and the setup() and loop() code can call `pass()` etc. without the `Test::pass()` since the test context is known.

In the current implementation, the summary is not reported until all active tests (including ones that are run repeatedly) are completed (reach pass, fail or skip).

wmacevoy, 2017-07-28:
... I am suffering from a little API remorse now that I read what I wrote; Test should have been called Testing and TestOnce should have been called Test.  I wonder how many users use this at all?

matthijskooijman, 2017-08-17:
Sorry for the delay, I got sidetracked with other things for a while. I've just pushed the changes I made to your library, perhaps that makes it clearer what direction I'm going: https://github.com/matthijskooijman/arduinounit/commits/multiple-runs
Also, here's the sketch that uses those changes: https://github.com/3devo/AtTinyBootloader/blob/master/BootloaderTest/BootloaderTest.ino#L478-L574

Originally, I wanted to have more control over tests (e.g. do some setup, run a single test, do more setup, run all tests, like that) which the `testing()` approach doesn't give. Looking at the way things ended up now, using `testing()` might actually work better, though that still doesn't give me test summaries between test runs. To be more specific: I think I want a test to be able to pass and fail multiple times, but with different configuration or in different contexts.

It's probably helpful if I post some serial output from my current setup, but I won't be able to generate that until tomorrow.

matthijskooijman, 2017-08-18:
Here's the serial output I promised:

```
****************************
Starting fixed tests
****************************
Test #1
Random seed: 16807
current addr = 0x0
reset addr = 0x8
Test 010_general_call_reset passed.
Test 020_ack passed.
Test 030_protocol_version passed.
Test 040_not_set_i2c_address skipped.
Test 050_set_i2c_address skipped.
Test 060_reread_protocol_version passed.
Test 065_short_long_read passed.
Test 070_get_hardware_info passed.
Test 075_get_serial_number passed.
Test 080_crc_error passed.
Test 100_command_not_supported passed.
Test 110_power_up_display passed.
Test 120_write_flash passed.
Test 130_invalid_writes passed.
Test summary: 12 passed, 0 failed, and 2 skipped, out of 14 test(s).
****************************
Test #2
Random seed: 311865808
current addr = 0x8
reset addr = 0x8
Test 010_general_call_reset passed.
Test 020_ack passed.
Test 030_protocol_version passed.
Test 040_not_set_i2c_address skipped.
Test 050_set_i2c_address skipped.
Test 060_reread_protocol_version passed.
Test 065_short_long_read passed.
Test 070_get_hardware_info passed.
Test 075_get_serial_number passed.
Test 080_crc_error passed.
Test 100_command_not_supported passed.
Test 110_power_up_display passed.
Test 120_write_flash passed.
Test 130_invalid_writes passed.
Test summary: 12 passed, 0 failed, and 2 skipped, out of 14 test(s).
****************************
Test #3
Random seed: 532104823
current addr = 0x8
reset addr = 0x9
using repstart after read
Test 010_general_call_reset passed.
Test 020_ack passed.
Test 030_protocol_version passed.
Test 040_not_set_i2c_address skipped.
Test 050_set_i2c_address skipped.
Test 060_reread_protocol_version passed.
Test 065_short_long_read passed.
Test 070_get_hardware_info passed.
Test 075_get_serial_number passed.
Test 080_crc_error passed.
Test 100_command_not_supported passed.
Test 110_power_up_display passed.
Test 120_write_flash passed.
Test 130_invalid_writes passed.
Test summary: 12 passed, 0 failed, and 2 skipped, out of 14 test(s).
```

For some extra context: This testcode runs on one Arduino-ish board, talking to an attiny over I²C that runs a bootloader. The bootloader is the software being tested. What happens is that the same series of tests runs repeatedly, but with different configuration to make it behave differently. After each series of tests, I now get a summary, and once a series does not complete succesfully, testing stops.

wmacevoy, 2017-08-20:
There is the ability to create your own summary with `Test::getCurrentX()`, where X = Skipped, Passed, Failed, or Count.  I decided to not put summary in a separate method because I wanted to keep the API small and the output generation is optional (it is possible to turn off that output so the code is not generated) and thought it would not be needed (your case excepted, but the getcurrent stuff lets you do that).

The multiple phases required thinking about how to remember the include/exclude patterns in the setup.  It was extra overhead most people would not use.  If you are willing to have an external source or flash control the state, you could first check which testing phase you are in and then configure the whole test suite that way.

... Looking at your pull, I see there is a code improvement for typeof ... could you make that a separate pull I can integrate ... I think it would just be better.

I also see there is a missing implementation feature that is making this easier to do than I thought.  I seemed to have skipped removing resolved tests from the static lists of tests.   This is a performance gain for those (vast majority) of cases that tests once resolved should not be part of the loop at all.  It also makes writing reset() much harder.

If you can make testing() work, that would be great.

matthijskooijman, 2017-08-30:
> ... Looking at your pull, I see there is a code improvement for typeof ... could you make that a separate pull I can integrate ... I think it would just be better.

Ok, done. I've also created a PR for the on_failure hook I added, which probably deserves some separate discussion.

As for converting my code to use `testing()`, there is still the problem that once a test is passed, failed or skipped, it is removed from the list of tests. I could use `testing()`, never pass tests (meaning that all tests that are not skipped or failed are implicitely passed) and then generate my own summary, but then I still cannot skip tests (since skipping a test removes it from the list, which prevents it from being run a second time). It wouldn't be terrible to not mark tests as skipped but instead just show them as passed in the summary, but that isn't very elegant.

I wonder if my suggested changes (adding keep_completed_tests option and Test::reset()) aren't a more elegant way to handle this. Is there anything you do (not) like about these changes?

wmacevoy, 2018-04-04:
I'm closing this with a "I don't know how to support this feature and maintain the simplicity of using the framework."

matthijskooijman, 2018-04-13:
I think I would like to revisit this issue again, ideally I can run my tests with an unmodified ArduinoUnit :-)

First off: since I originally reported this issue, my testsuite has become a bit simpler: It turns out that running individual tests is never really required. What I basically need is to run all tests until they are done, then change the environment, and then reset all tests so they can be run again.

In my original code, the `keep_completed_tests` flag I added was not so nice. It is annoying to need such a setting in the first place, and keeping completed tests in the list potentially complicates handling them. However, this morning I realized another option: Instead of throwing away tests after completion, you can also move them into a separate "done" list. This has negligable overhead, allows moving tests back from done into the "root" list later, and doesn't change the way the "root" list works.

I did a quick implementation of this here: https://github.com/matthijskooijman/arduinounit/commits/multiple-runs

Basically you can now run tests as normal, and once all are completed, you can call `resetAllTests()` to reset the state to what it was on startup, and include/exclude and run tests again.

 - When a test is skipped or excluded it is only marked as `DONE_SKIP`, it will not be moved into the `done` list until `Test::run()` sees it. This could possibly be done immediately, though some additional  care is required when this happens *during* `Test::run()`. If this is done immediately, then `include()` only needs to look at the `done` list as well.
 - With a `done` list, the `root` variable should probably be renamed to `active` (or `active_list` and `done_list` to make it more explicit that these are list heads?)
 - The proposed version of `resetAllTests()` only works when all tests are done, otherwise the counters are messed up. This should either be enforced, or some additional work is needed to guarantee the counters stay correct.
 - Similarly, if `resetAllTests()` is called when there are still tests in the `root` list that are marked as `DONE_SKIP`, then these are not reset, which might be unexpected.
 - I removed the `reset()` virtual method, since it didn't really seem necessary. When the test is run for the second time, its `setup()` is run again and if a test really needs to know it is running for the second time, it can keep a flag?

> I tried making a branch that had consistent reset() semantics that would not make the rest of (I think much more common) use cases confusing. 

Would your troubles still apply with the changes I propose now?

wmacevoy, 2018-04-15:
I had a similar implementation.  I think moving between lists at the end of the loop is better, because I have test that use outcomes to decide outcomes (pass if this fails), and shuffling that list sounds like a ordering specificity that I don't want to introduce.  My real pause was trying to explain what it did; I did not think about only allowing it after the summary phase.  Can you try to explain your semantics in the README and provide an example?  Adding one global for the discard pile is very low overhead for a reset....

matthijskooijman, 2018-05-17:
I had another look at this, and update my implementation and added a README section with an example at https://github.com/matthijskooijman/arduinounit/commits/multiple-runs

 - I changed the function from `resetAllTests()` to `resetDoneTests()`. Normally, you would call this function when all tests are done, but if there are still pending tests, these are just left untouched, which simplifies semantics.
 - I am no longer touching `Test::count`, so the statistics bookkeeping is simpler.
 - I added a `Test::runUntilDone()`, which is just a convenience helper that makes it easier to run the complete set of tests a few times.
 - There is still some fuzz wrt excluded tests, which might still be in the `root` list (if `Test::run()` has not ran yet), or already moved to the `done` list. Only the latter will be affected by `resetDoneTests()`, not the former (but this is nothing new: even without my changes only the former will be affected by `include`, not the latter). This is probably not much of an issue, though, since you'd typically run at least once between including/excluding and `resetDoneTests()`.

> I think moving between lists at the end of the loop is better, because I have test that use outcomes to decide outcomes (pass if this fails), and shuffling that list sounds like a ordering specificity that I don't want to introduce.

I'm not sure what you mean here? Do you mean moving tests from `root` to `done` only *after* the loop running all tests in `Test::run()`? If so, that would mean `Test::run()` has to go over the list twice. It would also mean that calling `resetDoneTests()` from inside a test would no longer mess up the list (which is currently forbidden but not enforced). As for the ordering, the tests are resorted when moving from `done` to `root`, so I think this is not an issue?

> Can you try to explain your semantics in the README and provide an example?

Done, see the last commit linked above.

matthijskooijman, 2018-08-15:
@wmacevoy, have you got a chance to look at my proposal yet? I just rebased it on top of current master and created a PR for easier review.

wmacevoy, 2018-08-28:
Sorry school is in so these will be slow.  Did you see my PR comment?  I'm trying to be consistent with a plan for continuous integration.

matthijskooijman, 2018-08-28:
I did, and commented on it this morning?

PR #85 (2023-05-10): Allow running tests multiple times
This PR adds support to run tests (typically the whole test suite) multiple times, by resetting tests that are marked as done. It also adds a convenience helper function `Test::runUntilDone()` which repeatedly runs any pending tests until all of them are done.

This fixes #61.
closed 2023-05-10 by wmacevoy