mmurdoch/arduinounit #68: Add support for ESP8266

ssilverman, 2017-11-11:
When compiling for 
ArduinoUnit/src/ArduinoUnitUtility/Compare.h:3:26: fatal error: avr/pgmspace.h: No such file or directory.

See [esp8266/Arduino/doc/faq/a03-library-does-not-work.rst](https://github.com/esp8266/Arduino/blob/d7044eceab4d4453e4d73ac49dcbbc8b8d0c9eb1/doc/faq/a03-library-does-not-work.rst)

ssilverman, 2017-11-11:
Ah, this is already in pull requests #57 and #58.

wmacevoy, 2018-04-04:
[thread summary] ArduinoUnit 2.3.5-alpha now supports flash string de-duplication on AVR and uses RAM on other platforms, along with a number of memory-saving optimizations that do not change the outcome of tests. It also supports ESP 8266 and ESP 32 platforms (along with development platform "en vitro" tests). Please try the latest pre-release and provide feedback on #73 if you are interested in these features.

closed 2018-04-04 by wmacevoy