mmurdoch/arduinounit #71: 'AUnit' library: a rewrite of ArduinoUnit to support ESP8266 and reduce flash memory usage

bxparks, 2018-03-13:
Hi everyone,

I just uploaded a rewrite of ArduinoUnit called AUnit to:
https://github.com/bxparks/AUnit
It is almost a drop-in replacement for ArduinoUnit.

AUnit solves 2 pressing problems for me:
* it works with ESP8266 (#68, #57, #55)
* it has a far smaller flash memory footprint (#70). On the unit test referenced in #70, AUnit is 66% smaller, in other words, it's 1/3 the size of ArduinoUnit.

I would like to thank the ArduinoUnit contributors for creating such an easy to use framework. I was hesitant to create a new derivative work, but ArduinoUnit has not seen a release in over a year, and I wanted to make more rapid progress.

Let me know if you have any questions, I'm happy to answer them.

wmacevoy, 2018-03-17:
Thanks for your comments; and your version of AU.  I am concerned about the fracturing of the API and the community that support it.  There is an pull request that remains API compatible that saves significant memory #70 --- I was planning on putting in that update as a default but configurable change.  Are you amenable to this?

I did ask for support of the ESP8266 and ESP32 chips; I didn't want to incorporate changes, without any commitment of support. 

Since it would be less work to support 1 framework with 1 api, would you be willing to become a contributor?  We could use the help...

bxparks, 2018-03-17:
Hi Warren,

I understand your concern about the split. But I think I'm pretty much committed to maintaining and extending my AUnit version. But I'm willing to help improve and test ArduinoUnit (even if only to learn and document incompatibility problems between the two libraries).

My impression was that development on ArduinoUnit was pretty much dead. The first report of ESP8266 problems was filed 14 months ago in #54, and it is still broken. Even something as simple as releasing a new version of `keywords.txt` that fixed the highlighting problem 7 months ago would have helped.

With regards to the flash memory usage problem of #70, my AUnit version consumes 66% less flash memory than ArduinoUnit. The best I could get ArduinoUnit was about a 50% savings. On a test that now consumes 54kB on ArduinoUnit, that extra 8.6kB is important to me.

With regards to ESP support, my understanding was that the `typeof()` function that produces the error in #54 is a GCC extension, not standard C++. I couldn't make much sense out of that error. It's not even clear to me why that `typeof()` is required in that expression. My AUnit library avoids this problem by not using the templating specialization. That kills 2 problems, less flash memory and works on ESP.

If you'd like me to take a look at some of your fixes, I'd be happy to. I've probably gotten to know the internals of ArduinoUnit better than the average user while I was rewriting it.

Brian

wmacevoy, 2018-03-17:
I seem to have time to work on issues for ArduinoUnit in the summers, which is the reason for the long delay.  In your opinion most of the flash is due to string replication (I see something that might fix that, I want to try it tonight) and template specialization?  What did you feel were the most memory intensive problems?

bxparks, 2018-03-18:
Looking at the table in #70, there's a 35% savings from removing `F(#arg1)` and `F(#arg2)`. There's a further 15% savings from changing the `F()` strings to `const char[]` strings in the `assertion()` method. I believe AUnit gets another 15% savings by avoiding all the duplicated code caused by template specialization of ArduinoUnit. It supports type mixing in assertion<A, B>(). I made an decision to support only assertion<A, A>() in AUnit, except for supporting the combinatorial mixing of the 3 string types.

wmacevoy, 2018-04-04:
[thread summary] ArduinoUnit 2.3.5-alpha now supports flash string de-duplication on AVR and uses RAM on other platforms, along with a number of memory-saving optimizations that do not change the outcome of tests.  It also supports ESP 8266 and ESP 32 platforms (along with development platform "en vitro" tests).  Please try the latest pre-release and provide feedback on #73 if you are interested in these features.

closed 2018-04-04 by wmacevoy