mmurdoch/arduinounit #72: Test::exclude isn't working properly. Test::exclude("*");

gtame, 2018-03-17:
Hi ,
if the first exclude sentence is  Test::exclude("*").  The sentence doesn't work.

[image]

`
#include "main.h"
 
test(correct)
{
	int x=1;
	assertEqual(x,1);
}

test(incorrect)
{
	int x=1;
	assertNotEqual(x,1);
}

test(otro)
{
	int x=1;
	assertNotEqual(x,1);
}

extern "C" void __cxa_pure_virtual() {}

int main() {

	init();
	setup();
         while (true)
	loop();
}

void setup()
{
	Serial.begin(9600);
	while(!Serial)
	
	Test::exclude("*");
	Test::include("incorrect");
}

void loop() 
{ 
	Test::run();
}
`

However if I added before another exclude sentence works. Ej, Test::exclude("*e")
`
void setup()
{
	Serial.begin(9600);
	while(!Serial)

 	Test::exclude("*e");
	Test::exclude("*");
	Test::include("incorrect");
 
}
`
[image]

wmacevoy, 2018-03-18:
There needs to be a ';' after the `while (!Serial)`, perhaps preferably `while (!Serial) { /* busy */ }` -- the line after the while (exclude) was never executed.  Pretty subtle; sorry C syntax got you there...

closed 2018-03-18 by wmacevoy

gtame, 2018-03-19:
Hi, it's clear. A tree didn't let me to see the forest😀