mmurdoch/arduinounit #75: mock

wmacevoy, 2018-03-29:


wmacevoy, 2018-03-29:
I ported WString.h, WString.cpp, Print.h, Print.cpp, Stream.h and Stream.cpp to mocked versions in the mock branch.  Now I need a license reassignment....

matthijskooijman, 2018-03-29:
In a private e-mail, @wmacevoy just asked the contributors to these files (including me) to consider relicensing to MIT. Since it seems better to discuss this in the open, I'm replying here.

First off: I'm happy to relicense my contributions to MIT. More specifically, I hereby offer my contributions to Arduino's WString.h, WString.cpp, Print.h, Print.cpp, Stream.h and Stream.cpp under the MIT license.

I do wonder if this is really needed? I guess the mock code (or perhaps even each of these three source/header combo's) could be seen as a separate library (though I'm not entirely sure if the LGPL terms allow this).

Regardless, LGPL and MIT are actually compatible licenses, so mixing code under both licenses should not be problematic. If the LGPL code is not seen as a separate library, I believe the viral aspect of LGPL means that any compiled *binary* must be licensed under the LGPL license (and probably dual-licensed under MIT too), but no such restriction is on the source code AFAIK.

wmacevoy, 2018-03-29:
The private email was to direct discussion here; I don't know how else to reach Arduino contributors.

You can deliver an application with both MIT and LGPL components, but that promotes the whole project to LGPL (viral clause and AU is a library).  Using AU with LGPL would force all of AU to become LGPL.  Considering how many forks of AU there are out there, I’m sure this would be problematic.

matthijskooijman, 2018-03-29:
> You can deliver an application with both MIT and LGPL components, but that promotes the whole project to LGPL

It indeed seems this is not just for a binary distribution (as I previously suggested), but also for a source distribution (if the resulting source is a "work based on the [GPL] Program" ([GPL3 section 5](https://www.gnu.org/licenses/gpl-3.0.html)).

However, this would mean that you must make the whole available under LGPL, but there is nothing that prevents you from *also* making it available as MIT. Also, I believe that the MIT license (granted by you and your contributors) does not prevent this dual-licensing (which is what makes MIT "GPL-compatible").

All of the above are the terms of the GPL, which only applies if the LGPL exception does not apply, of which I'm not convinced.

Just the fact that you're distributing the LGPL parts and your own code in the same repository, does not, IMHO, change the relation between these files neccesarily. Normally, when compiling under Arduino, these files are also included in the link and it seems obvious that the LGPL exception applies. Then why not if you put these files in your own  repository? It seems some degree of separation would be needed (e.g. a separate "mock" directory or something), but even just separate files might be sufficient (IMHO).

Looking at the definitions in the [LGPL3](https://www.gnu.org/licenses/lgpl-3.0.en.html):

> An “Application” is any work that makes use of an interface provided by the Library, but which is not otherwise based on the Library. Defining a subclass of a class defined by the Library is deemed a mode of using an interface provided by the Library.

It seems that this would apply to your code (though "based on" is not really defined by these licenses). None of these definitions assign any additional copyright-coupling based on how closely code is distributed together AFAICS.

If the LGPL exception would apply, then the notable requirements it poses is to carry notices (e.g. in the README), and to allow relinking with a modified implementation (which is automatically satisfied because you are distributing source, even if not under the LGPL).

wmacevoy, 2018-03-29:
I disagree.  Unless you have a more permissive license for the source, you cannot derive work from GPL/LGPL without invoking the viral clause.  Having an LGPL mock library would solve this for AU, but would be very problematic for any developer using it.

matthijskooijman, 2018-03-29:
It's your work, so your call :-)

Like I said, I'm happy to relicense, so let's hope others do too.

@cmaglie, IIRC these files are the ones that were being considered to be moved out of the Arduino cores into some library or otherwise shared between the different Arduino architectures (these files are now duplicated). Were there any licensing considerations for that effort?

wmacevoy, 2018-04-02:
Thanks for allowing the reassignment, but I haven't heard from David A. Mellis, who is on every copyright.  I have rewritten WString and Print.  Stream is more of a pain; but not in the core of AU.

wmacevoy, 2018-04-03:
Ok Stream is now mocked as a re-write. The entire AU library, including mocks, is MIT.  Thanks everyone for offering your changes, but this was an all-or-nothing ask.

closed 2018-04-03 by wmacevoy