mmurdoch/arduinounit #77: assertXxx() produces unexpected and sometimes incorrect results for mixed integer types on 8-bit and 32-bit processors

bxparks, 2018-04-04:
# Summary

The following problems were observed in latest code on `master`. Some (maybe all) of these problems were also observed in 2.2.0, but I don't recall if the behavior was exactly the same. The code shown below was tested on an Arduino Nano (ATmega328P), a Teensy LC (ARM Cortex-M0+, 32-bit), and an NodeMCU v1.0 (ESP8266, 32-bit).

**Problem 1**: With the default compiler warning settings, the compiler warnings about integer type mismatch are not printed. The `assertXxx()` sometimes produces unexpected results for non-portable code. 

*Expected result*: A compile time failure would be preferable, since assert statements with mixed integer types are often non-portable and doesn't do what the user actually intended.

**Problem 2**: `assertLess()` produces an incorrect result on 32-bit processor.

*Expected result*: `assertLess()` should behave exactly like its equivalent `if-statement` counterpart on a 32-bit processor.

# Preamble

In the revised templatized code in 2.3.x, the various `assertXxx()` macros eventually evaluate to the templatized `assertion()` method:
```
template<typename A, typename B, typename F>
static bool assertion(...) { ... }
```
which takes an instance of one of the templatized `compareXxx()` methods in:
```
template<typename A, typename B> struct Compare {
  static bool equal(const A&, const B&);
  static bool ...;
  ...
}
```

The problem occurs when the type of `A` and `B` are mixed signed and unsigned integers, and of different sizes.

# Problems

# Problem 1

With the compiler warning setting at "Default" (which I believe is the default), there is no warning printed about comparing signed and unsigned integers, which may not do what the user intended. At the "All" setting, we get the following, but I think most people would not see these warnings:
```
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/Compare.h: In instantiation of 'static bool Compare<A, B>::less(const A&, const B&) [with A = int; B = unsigned int]':
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/Compare.h:40:173:   required from 'bool compareLess(const A&, const B&) [with A = int; B = unsigned int]'
sketch.ino:68:3:   required from here
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/Compare.h:21:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return a<b;
```

The following is the sample test case that generates this warning, which is hidden for most users. In this case, we have a test that passes on (8-bit) AVR, but fails on a 32-bit processor. You could argue that it's not ArduinoUnit's fault that the user wrote non-portable code. That's true, but I also think that dev tools should help users detect such conditions, and the fact that ArduinoUnit  accepts this code (silently with the default "Default" compiler warnings setting) is somewhat unfortunate.

```
#if defined(__AVR__)

test(case_1_intended_to_pass_and_passes_on_avr)
{
  verbosity |= TEST_VERBOSITY_ASSERTIONS_PASSED;

  uint16_t end = 1;
  uint16_t start = 3;
  int limit = 3;

  // Two's complement rollover is intentional, taking advantage of 16-bit int on
  // AVR. The assertMore() passes and does the right thing, but the code is
  // non-portable and brittle, for example, we see that it fails on a 32-bit
  // processor.
  if (end - start > limit) {
    Serial.println("The following assertMore() should pass on AVR");
  } else {
    Serial.println("The following assertMore() should fail on AVR");
  }
  assertMore(end - start, limit);
}

#elif defined(__arm__) || defined(ESP8266)

[... 98 lines cut ...]

On an AVR, this produces the following, which is expected, with a caveat:
```
The following assertLess() should fail on AVR
Assertion failed: (end - start=-1) < (limit=30), file sketch.ino, line 22.
Test case_2_nonportable_code_intended_to_pass_but_fails_on_avr failed.
Test summary: 0 passed, 1 failed, and 0 skipped, out of 1 test(s).
```
The assertion result is correct (failed), but the assertion message is **misleading**, because it says that `(end - start=-1) < (limit=30)` which is actually a `true` expression and should have passed. (My guess is that the there was an internal conversion to unsigned int, which caused the expression to be `false`, even through it's printed to be `true`.)

On ARM and ESP8266, the `assertLess()` fails, which is **unexpected and I think incorrect** result, because the if-statement evaluates to `true`, so I would expect that the `assertLess()` should evaluate to `true`, but it fails and produces this result:
```
The following assertLess() should pass on 32-bit MCUs
Assertion failed: (end - start=-1) < (limit=30), file sketch.ino, line 45.
Test case_2_intended_to_pass_and_should_pass_on_32_bit failed.
Test summary: 0 passed, 1 failed, and 0 skipped, out of 1 test(s).
```

# Comments

I suppose we could suggest to developers to avoid mixing signed and unsigned types, and mixing different integer sizes. I'm not sure that's practical. Maybe that suggestion works in a desktop or server environment, but in an embedded environment, I find myself using a variety of integer sizes to maximize static memory efficiency. And I find myself taking advantage of 2's complement unsigned rollovers quite often, to avoid having to use a larger integer type.

I also think the problems discussed here point to the advantages and necessity of running unit tests on the embedded environment itself, over running the tests on mocked out desktop or cloud environment. We can't really be sure something works until it compiles and runs on the target embedded environment.

Quick aside: I made the deliberate decision to try to avoid these problems in [AUnit](https://github.com/bxparks/AUnit) by requiring the two arguments in `assertXxx()` to be identical (except for "string" types). AUnit produces a compile-time error (which unfortunately is somewhat lengthy and obtuse) if the types are not identical. I'm not saying that this was a better design choice, but just wanted to point out an alternative.

bxparks, 2018-04-04:
One bit of context that I left out in my original post which might help you digest all this. I was trying to create 2 example tests:

* case 1: one which succeeds (intentionally) on AVR but fails on 32-bit environments, and
* case 2: one which succeeds (unintentionally) on 32-bit environments, but fails on AVR

I was successful in case 1. I'm pretty sure that I succeeded in case 2, because the if-statement does what I want, but that's when I found that `assertLess()` doesn't do what the if-statement does.

Case 2 is where mock unit tests running on some other (32-bit) environments will have trouble detecting problems with code that will fail on AVR.

wmacevoy, 2018-04-04:
The 2.3 branch does type widening to limit the number of specializations of the comparison that are made.  char and short promote to int, unsigned char and unsigned short promote to unsigned, float promotes to double.  After promotion, both sizes are the same, so the C++ rule is to compare as  unsigned. I did this because it saves a lot of specializations (and therefore space) and did not realize it changed comparison semantics (the compiler warns you if you have those options enabled as you pointed out) between signed and unsigned types.  I think most of the space saving comes from unifying strings, so maybe just putting back the integer type specializations would get back to a<b equivalency (on the same hardware).  Getting a<b equivalency across platforms sounds like a pipe dream, and the argument for allowing unit tests in the embedded environment in any case.  I created a branch iss77 which removes the extra widening.  Hard call.  The compiler does generate warnings.  The extra space.

bxparks, 2018-04-05:
So the latest version on master does not compile on Teensy or ESP8266. The error messages are about the same on the 2 platforms, modulo file paths. I think I mentioned previously that you don't need to own these microcontrollers to verify, you can verify the compile without uploading.

```
Build options changed, rebuilding all
In file included from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/WString.h:28:0,
                 from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/Print.h:38,
                 from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/Stream.h:24,
                 from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/HardwareSerial.h:252,
                 from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/WProgram.h:46,
                 from /home/brian/Downloads/Arduino/arduino-1.8.5/hardware/teensy/avr/cores/teensy3/Arduino.h:3,
                 from /tmp/arduino_build_983557/sketch/ArduinoUnitCompareProblem.ino.cpp:1:
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:30:33: error: expected nested-name-specifier before numeric constant
 template <typename _A, typename _B> struct ArduinoUnitDoubleWide {
                                 ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:30:33: error: expected '>' before numeric constant
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:32:11: error: expected unqualified-id before numeric constant
   typedef _B B;
           ^
In file included from /home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/Compare.h:3:0,
                 from /home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnit.h:12,
                 from sketch.ino:2:
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:37:57: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < char , char > {
                                                         ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:37:57: note:   expected a constant of type 'int', got 'char'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:42:58: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < char , short > {
                                                          ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:42:58: note:   expected a constant of type 'int', got 'short int'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:47:56: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < char , int > {
                                                        ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:47:56: note:   expected a constant of type 'int', got 'int'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:52:57: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < char , long > {
                                                         ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:52:57: note:   expected a constant of type 'int', got 'long int'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:57:62: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < char , long long > {
                                                              ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:57:62: note:   expected a constant of type 'int', got 'long long int'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:62:58: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < short , char > {
                                                          ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:62:58: note:   expected a constant of type 'int', got 'char'
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:67:59: error: type/value mismatch at argument 2 in template parameter list for 'template<class _A, int <anonymous> > struct ArduinoUnitDoubleWide'
 template < > struct ArduinoUnitDoubleWide < short , short > {
                                                           ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnitUtility/ArduinoUnitWiden.h:67:59: note:   expected a constant of type 'int', got 'short int'
[... cut ...]
 #define ArduinoUnitMacroChoose3(f,...) ArduinoUnitMacroRecompose3((null,##__VA_ARGS__,f ## 3, f ## 2,f ## 1, f ## 0,))
                                        ^
/home/brian/dev/arduino/libraries/ArduinoUnit/src/ArduinoUnit.h:598:25: note: in expansion of macro 'ArduinoUnitMacroChoose3'
 #define assertMore(...) ArduinoUnitMacroChoose3(assertMore_, ## __VA_ARGS__)(__VA_ARGS__)
                         ^
sketch.ino:45:3: note: in expansion of macro 'assertMore'
Error compiling for board Teensy LC.
```

wmacevoy, 2018-04-06:
Ok, didn't foresee that.  I have some automation of builds now for uno/mega/esp8266.  How do I build for teensy?  I don't see that as an option...  Try the latest in this branch....

closed 2018-04-11 by wmacevoy