mmurdoch/arduinounit #82: Mocking I/o

T81, 2018-06-11:


wmacevoy, 2018-06-11:
ArduinoUnit does not provide that support directly.  How are your running ArduinoUnit?  If you are "en vitro" then you can provide your own.  Simavr (simduino) can provide digital signals.  "en vivo" (on the board) requires a bit of refactoring (of the code you are testing) to test with mocks.  Can you explain more?

T81, 2018-06-13:
I am writing a library for driving motors. At some point I am reading some digital inputs and analog inputs. I would like to mock these inputs to certain values in some way if possible. I do not know if I made this clear enough. I'll try to give a better example tomorrow. I'll also check simduino.

wmacevoy, 2018-06-14:
https://gist.github.com/wmacevoy/7ef9468ff95f3d8d459f48f2e76029e1

Might be helpful.  If you are willing to have some runtime overhead, mocking is usually done with polymorphism, if that means anything to you.  The template style make the real implementation not have any overhead, which is better in many embedded situations...

T81, 2018-06-15:
Thank you very much! I'll have feed back in a couple of weeks or so. Also I will upload the library and the unit tests to GitHub. Currently I am working with the wiki: https://github.com/L298/wiki
You can find the unit tests here: https://gist.github.com/T81/fbe0a2f0d61f9fb7c67bf5fa384b3545

closed 2019-09-14 by wmacevoy