mmurdoch/arduinounit #86: V3.0.2: serial message for each test is "skipped"

ttttttom, 2018-12-04:
I tested arduinounit Version 2.2.0:
Messages are ok.

Same program with arduinounit Version 3.0.2:
Messages always say test skipped for each test. Test-Summary is correct though.

ttttttom, 2018-12-05:
Software:
- Arduino IDE V1.8.5
- ArduinoUnit V3.0.2

Hardware: 
- BN: Arduino/Genuino Uno
- VID: 2341
- PID: 0043
- SN: 55437333637351208071

For the example sketch "basic.ino" the output is:
> Test correct skipped.
> Assertion failed: (x=1) != (1=1), file basic.ino, line 13.
> Test incorrect skipped.
> Test summary: 1 passed, 1 failed, and 0 skipped, out of 2 test(s).

`#line 2 "basic.ino"
#include <ArduinoUnit.h>
test(correct)
{
  int x=1;
  assertEqual(x,1);
}

test(incorrect)
{
  int x=1;
  assertNotEqual(x,1);
}

void setup()
{
  Serial.begin(9600);
  while(!Serial) {} // Portability for Leonardo/Micro
}

void loop()
{
  Test::run();
}`

wmacevoy, 2018-12-14:
I cannot reproduce this.  I built on a Mega 2560 & Uno R3 on Arduino 1.8.5 (OS X) and got the expected output:

```
Test correct passed.
Assertion failed: (x=1) != (1=1), file basic.ino, line 12.
Test incorrect failed.
Test summary: 1 passed, 1 failed, and 0 skipped, out of 2 test(s).
```

Could you try reinstalling the library, trying another board, and/or reinstalling the arduino IDE?  It's pretty surprising that your result is so different, but I don't know how to help.

ttttttom, 2019-01-05:
I think I can give you a hint now. Please check your "Arduino AVR Boards" Version in the board manager. On my system (Windows with Arduino 1.8.1, Nano old boot loader & UnoR3):
- Using Arduino AVR Boards V1.6.23 with Arduino Unit 3.0.2 -> wrong unit test output
- Using Arduino AVR Boards V1.6.22 with Arduino Unit 3.0.2 -> wrong unit test output
- Using Arduino AVR Boards V1.6.21 with Arduino Unit 3.0.2 -> correct unit test output
and
- Using Arduino AVR Boards V1.6.23 with Arduino Unit 2.2.0 -> correct unit test output

I hope this will help you...

ttttttom, 2019-01-05:
Checked it also on Linux and OSX with Arduino 1.8.8 and 1.8.5. Always the same result ...

wmacevoy, 2019-02-17:
Ok found a hint: https://github.com/arduino/ArduinoCore-avr/issues/7

1.6.21 was the last version created from the IDE repo, and also the last one to require `gcc-4.9.2-atmel3.5.4-arduino2` (I think this last information might be the one you need most)

commit 9dbd3cec (2019-09-09): resolve #86 - finally!
closed 2019-09-09 by wmacevoy