mmurdoch/arduinounit #87: Doesn't compile for Teensy 3.2

ssilverman, 2019-02-26:
The error message:
> src/ArduinoUnitUtility/ArduinoUnitMockPrint.cpp:136:95: error: no matching function for call to 'MockPrint::concat(const char*, size_t&)'

ssilverman, 2019-03-04:
My fix in `ArduinoUnitMockPrint.cpp`:

```c++
size_t MockPrint::write(const uint8_t *buffer, size_t size) {
#ifdef TEENSYDUINO
  append((const char *)buffer, size);
#else
  concat((const char *)buffer, size);
#endif
  return size;
}
```

wmacevoy, 2019-03-06:
I'm looking at the teensy source,

https://github.com/PaulStoffregen/cores/blob/master/teensy/Print.h

And see no concat.  What method are you mocking?

wmacevoy, 2019-03-06:
Sorry took a bit to think about what this was doing.  Can you make a Pull Request & I will encorporate your change?

ssilverman, 2019-03-17:
#88

dogweather, 2019-09-07:
@wmacevoy Thanks for merging this! Would you consider cutting a new release so we can use it easily in the Arduino IDE?

wmacevoy, 2019-09-12:
3.0.3+ has the merge.

closed 2019-09-12 by wmacevoy